japaric/stm32f103xx-hal #16: Missing dependencies for examples and linker-error after trying to fix them

ChristianRinn, 2017-06-29:
Hi and first of all thank you for all the effort that you put into making rust run on small microcontrollers.

Out of curiosity I tried to build the blinky example by simply cloning the repo and running

`xargo build --target thumbv7m-none-eabi --example blinky`

The problem is that the dependencies for the cortex-m-rt and cortex-m-rtfm crates seem to be missing in the Cargo.toml which is probably why I get an error (see below).

Furthermore the dependency for compiler_builtins seems to be missing in the cortex-m-rt crate.

I tried to fix the missing dependencies myself and could get it to compile but after all I got a linker error because of some undefined references https://gist.github.com/ChristianRinn/2b57a10bbe6e9d41d26d1169e7cbb064

This linker error could very well be because I handled the compiler_builtins dependency wrong. Maybe you have an idea what is going wrong.

Regarding my toolchain:
I'm using Arch Linux with arm-none-eabi-newlib-2.5.0.20170421-1 and arm-none-eabi-gcc-7.1.0-1, i.e.
arm-none-eabi-gcc (Arch Repository) 7.1.0 
and furthermore
rustc 1.20.0-nightly (445077963 2017-06-20)
xargo 0.3.8
cargo 0.21.0-nightly (50b1c24d1 2017-06-17)

------------------------------------------
Dependency error:
    Updating registry `https://github.com/rust-lang/crates.io-index`
    Updating git repository `https://github.com/japaric/embedded-hal`
    Updating git repository `https://github.com/japaric/nb`
   Compiling vcell v0.1.0
   Compiling aligned v0.1.1
   Compiling either v1.1.0
   Compiling cast v0.2.2
   Compiling cortex-m-semihosting v0.1.3
   Compiling static-ref v0.1.1
   Compiling nb v0.1.0 (https://github.com/japaric/nb#4696bd09)
   Compiling blue-pill v0.1.0 (file:///home/james/tmp/blue-pill-blinky)
   Compiling cortex-m v0.2.11
   Compiling volatile-register v0.2.0
   Compiling embedded-hal v0.1.0 (https://github.com/japaric/embedded-hal?rev=5295697669f5b48a900aa325b8ebb4d4e8d4b236#52956976)
   Compiling stm32f103xx v0.6.1
error[E0463]: can't find crate for `cortex_m_rt`
  --> examples/blinky.rs:12:1
   |
12 | extern crate cortex_m_rt;
   | ^^^^^^^^^^^^^^^^^^^^^^^^^ can't find crate

error: aborting due to previous error(s)

error: Could not compile `blue-pill`.

To learn more, run the command again with --verbose.

japaric, 2017-06-29:
[You have to start from the cortex-m-quickstart template and add this crate as a dependency](https://japaric.github.io/blue-pill/blue_pill/#usage) (as well as the cortex-m-rtfm crate). Then you can copy over the examples from this repo to the quickstart project.

ChristianRinn, 2017-07-04:
Thank you Jorge, that was too easy and my first rusty blinky is working now. I always thought that examples provided with a crate would be able to compile on their own. Thank you again for your awesome work with Rust on microcontrollers.

closed 2017-07-04 by ChristianRinn