japaric/stm32f103xx-hal #17: error: linking with `arm-none-eabi-gcc` failed: exit code: 1

HemantJoshi, 2017-07-02:
Copied Xargo.toml (from cortex-m-quickstart) into blue-pill folder.
Added missing crates/dependencies in to the Cargo.toml.
The compilation was through. However gave linking problem. Please see the log.
$ xargo build --target thumbv7m-none-eabi --example blinky
Compiling nb v0.1.0 (https://github.com/japaric/nb#4696bd09)
   Compiling static-ref v0.1.1
   Compiling quote v0.3.15
   Compiling vcell v0.1.0
   Compiling either v1.1.0
   Compiling cortex-m-semihosting v0.1.3
   Compiling unicode-xid v0.0.4
   Compiling typenum v1.9.0
   Compiling cortex-m-rt v0.2.4
   Compiling r0 v0.2.1
   Compiling cortex-m v0.2.11
   Compiling cast v0.2.2
   Compiling aligned v0.1.1
   Compiling volatile-register v0.2.0
   Compiling embedded-hal v0.1.0 (https://github.com/japaric/embedded-hal?rev=5295697669f5b48a900aa325b8ebb4d4e8d4b236#52956976)
   Compiling synom v0.11.3
   Compiling syn v0.11.11
   Compiling stm32f103xx v0.6.1
   Compiling cortex-m-rtfm v0.1.1
   Compiling blue-pill v0.1.0
error: linking with `arm-none-eabi-gcc` failed: exit code: 1
....
....
= note: /usr/bin/../lib/gcc/arm-none-eabi/6.3.1/../../../../arm-none-eabi/lib/crt0.o: In function `_start':
          (.text+0xe0): undefined reference to `__libc_init_array'
          /usr/bin/../lib/gcc/arm-none-eabi/6.3.1/../../../../arm-none-eabi/lib/crt0.o: In function `_start':
          (.text+0xf0): undefined reference to `exit'
...
...
error: aborting due to previous error(s)

error: Could not compile `blue-pill`.

$
// log ends

HemantJoshi, 2017-07-02:
Forgot to mention earlier that as per instruction removed build.rs and memory.x before compilation.

ChristianRinn, 2017-07-04:
I had similar problems in #16 but the solution is pretty simple as stated in https://japaric.github.io/blue-pill/blue_pill/#usage . I.e. you start with the cortex-m-quickstart crate and remove the build.rs and memory.x . Then you copy build.rs and memory.x from the blue-pill crate to the cortex-m-quickstart directory and copy one of the examples (e.g. blinky.rs) from the blue-pill example folder to the example folder in your cortex-m-quickstart directory. After all you should be able to compile with
`xargo build --target thumbv7m-none-eabi --example blinky`

After succesfully compiling and linking you can find your binary in target/thumbv7m-none-eabi/debug/examples

HemantJoshi, 2017-07-05:
Ok. So following is what I did to successfully build blinky.rs for blue-pill.
1. Cloned the cortex-m-quickstart.
2. Went inside cortex-m-quickstart
3. Removed memory.x and build.rs, because these two are coming from blue-pill crate.
4. Added blue-pill, embedded-hal and cortex-m-rtfm crates into Cargo.toml, the same way it was in the blue-pill Cargo.toml. 
5. Lastly copied blinky.rs from blue-pill.
6. Ran "xargo build --target thumbv7m-none-eabi --example blinky"

Notes:
1. The crates which are added into the Cargo.toml, are downloaded from github as per the specified URLs. However some might face problem with downloading https:, this is because your service provider's restrictions. In that case one should use git:.
2. The blue-pill crate encapsulates a few crates (for example embedded-hal, ..). So why should one need to add those crates separately into the app (blinky.rs)? There should be better way referring to those crates in blinky.rs, (say by first referring to blue-pill and recursively referring the crates). Of course this is a humble suggestion. With this, one doesn't have to modify Cargo.toml within cortex-m-quickstart which is a generic crate.

-Hemant

ChristianRinn, 2017-07-05:
Hello Hemant,
good to hear that you were able to compile one of the examples. I agree with you that it is somewhat counter intuitive that you have to copy over the examples, the build.rs and memory.x files from the blue-pill to the cortex-m-quickstart directory. Maybe @japaric can tell a little more about his design decisions.

HemantJoshi, 2017-07-05:
Thanks Christian.
I am exploring Embedded Rust for a week or so. I am impressed by the current status of cortex-m Rust, all done by JorgeAparicio and the collaborators. I think Rust has great future especially in Embedded domain.

japaric, 2017-07-05:
> that you have to copy over the examples, the build.rs and memory.x files from the blue-pill to the cortex-m-quickstart directory.

As @HemantJoshi said the build.rs and memory.x *don't* have to be copied over as they are provided by the blue-pill crate.

As for why `xargo build --example` doesn't work on this repo: that's because this is a *library* crate. As you have seen applications require some extra files like Xargo.toml and .cargo/config which are provided by cortex-m-quickstart. I'd prefer not to have a *copy* of those files here because that makes it possible for the copy to get out of date.

That being said, there are some changes in the pipeline (device crates will depend on cortex-m-rt) that would make it reasonable to add .cargo/config here, plus Xargo.toml may not be required in the future when `rustup target add thumbv7m-none-eabi` works.

> Even though blue-pill crate itself doesn't require any additional crate to be encapsulated within it, an app that uses blue-pill (say) blinky.rs does require some crates such as embedded-hal.

I think it makes sense to re-export the embedded-hal traits in blue-pill to avoid having the app depend on embedded-hal again. The other crates like cortex-m-rtfm are rather application specific so I don't know how much it would help to have a crate that re-export all those crates.

HemantJoshi, 2017-07-06:
Since the blinky worked for me, I am closing the issue.

closed 2017-07-06 by HemantJoshi

ChristianRinn, 2017-07-06:
> As @HemantJoshi said the build.rs and memory.x don't have to be copied over as they are provided by the blue-pill crate.

Thanks for making that clear to me, I wasn't aware of that but obviously it makes sense.

> As for why xargo build --example doesn't work on this repo: that's because this is a library crate. As you have seen applications require some extra files like Xargo.toml and .cargo/config which are [...]

Ok, now that makes sense to me as well although I think it would be really nice to be able to `git clone` a repo, run `xargo build --example` and start hacking without having to modify a Cargo.toml or delete files. Maybe an intermediate solution could be a blue-pill-examples repo which is just the cortex-m-quickstart repo with the appropriate Cargo.toml and the bluepill examples. On the other hand that is again one more thing to keep track off and as I see it at the moment you are really busy and moving on fast but I totally agree with Hemant

> I am impressed by the current status of cortex-m Rust

which is (as far as I can tell) mostly your accomplishment.