japaric/stm32f103xx-hal #21: SPI trait design

nikhilkalige, 2017-07-15:
The current SPI has all the pin it needs on the same GPIO (`PA` for `SPI`). However other SPI's on my board are spread across different GPIO's. So would it make sense to design the trait as below
```rust
pub unsafe trait SPI: Deref<Target = spi1::RegisterBlock> {
   fn init(&self);
}

unsafe impl SPI for SPI3 {
  fn init(&self) {
    gpioa.configure(); // has sclk and mosi
    gpiob.configure(); // has miso and ss
  }
}
```
_There may be some syntax issues in the above code._ 
I was unsure if this was valid as the `init` function for the SPI currently gets passed the peripheral as below
```rust
fn init(p: init::Peripherals) {
    let spi = Spi(p.SPI1);
    spi.init(p.AFIO, p.GPIOA, p.RCC);
}
```

With the design I have shown, the `spi.init` would have the following call `spi.init(p.AFIO, p.RCC)`. Would this cause issues with RTFM?

japaric, 2017-07-17:
>  So would it make sense to design the trait as below

The problem with this trait and implementation is missing `gpioa` and `gpiob` in the argument of `init`. Ideally you want to write the trait like this:

``` rust
pub unsafe trait SPI: Deref<Target = spi1::RegisterBlock> {
    type Peripherals;

    fn init(&self);
}

unsafe impl SPI for SPI3 {
    type Peripherals = (&'a GPIOA, &'a GPIOB);

    fn init(&self, p: (&GPIOA, &GPIOB)) {
        let (gpioa, gpioab) = p;

        gpioa.configure(); // has sclk and mosi
        gpiob.configure(); // has miso and ss
    }
}
```

but you can't because the `'a` in `type Peripherals` would be unbounded. A possible fix would be Associated Type Constructors (cf rust-lang/rfcs#1598) but the RFC has not been approved. Another fix would be to write `type Peripherals = (&'static GPIOA, &'static GPIOB)` but that would be limiting because the `'static` lifetime constraint is not required. In particular, RTFM doesn't hand out `&'static` references.

getreu, 2017-09-03:
Good news:

The Generic Associated Types RFC has been merged!

This will enable streaming iterators and a lot of other patterns.

[rfcs/1598-generic_associated_types.md at a7cd91048eea3d7ae83bec20446e62bad0c45381 · rust-lang/rfcs · GitHub](https://github.com/rust-lang/rfcs/blob/a7cd91048eea3d7ae83bec20446e62bad0c45381/text/1598-generic_associated_types.md)

If you want to learn more, there's a podcast episode about this RFC at [Request For Explanation #4 - Literally Haskell](https://request-for-explanation.github.io/podcast/ep4-literally-haskell/index.html)

japaric, 2017-09-21:
@nikhilkalige Thinking more about this. `init` doesn't need to be in the `Spi` trait. The current trend has been to *only* implement I/O actions as traits (read, write, etc.). And I don't think you'll need more than one GPIO to *do* I/O with SPI (the only GPIO you would need to drive would be "Slave Select" and some implementations of the SPI peripheral let you drive that pin using a SPI register), and that GPIO could be stored as a filed of the, e.g., Spi1 struct. For multi slave configurations I don't think it makes sense to be generic around all the "Slave Select" I/O lines; those are application specific and should be left up to the application writer -- they can write their own Spi wrappers to add Slave Select functionality. 

So, you could implement `init` individually on each Spi instance: `impl Spi1 { .. }`, `impl Spi2 { .. }`, etc.

@getreu But how long will it take to be *implemented* and become relatively bug free? :P

japaric, 2018-01-24:
The new SPI is pretty flexible about pin configuration thanks to the new ownership based approach to peripherals (see [blog post] for details about the new peripheral model).

[blog post]: http://blog.japaric.io/brave-new-io/

closed 2018-01-24 by japaric