japaric/stm32f103xx-hal #30: compiler error: panic_fmt

MrBuddyCasino, 2017-11-14:
When building the examples (usart1), I get: `error: language item required, but not found: panic_fmt`
I solved this with the following magic incantation: 

```
#[lang = "panic_fmt"]
#[no_mangle]
pub fn panic_fmt(_fmt: core::fmt::Arguments, _file_line: &(&'static str, u32)) -> !
{ loop { } }
```

And adding `#![feature(lang_items)]`. Not sure this is how its supposed to be? So far I get no UART output.

japaric, 2017-11-14:
>  I get: error: language item required, but not found: panic_fmt

That sounds rather bad.

Do you happen to have the [CARGO_INCREMENTAL](https://docs.rs/cortex-m-quickstart/0.2.1/cortex_m_quickstart/#used-cargo_incremental1) environment variable set? Because that's how you can get all sort of non-sensical compiler / linker errors.

Do you see this error with other examples?

MrBuddyCasino, 2017-11-14:
> Do you see this error with other examples?

Yes. I unset CARGO_INCREMENTAL, no change. Here are some versions in case that helps:

```
xargo --version
xargo 0.3.8
cargo 0.23.0-nightly (e447ac7e9 2017-09-27)
```

```
rustc -V
rustc 1.22.0-nightly (4c053db23 2017-10-22)
```

```
arm-none-eabi-ld -V
GNU ld (GNU Tools for ARM Embedded Processors 6-2017-q2-update) 2.28.0.20170620
```

Cargo dependencies:
```
[dependencies]
stm32f103xx = "0.7.5"
cortex-m-rtfm = "0.2.1"
#stm32f30x = "0.4.1"

[dependencies.blue-pill]
git = "https://github.com/japaric/blue-pill"

[dependencies.cortex-m]
version = "0.3.1"

[dependencies.cortex-m-rt]
features = ["abort-on-panic"]
git = "https://github.com/japaric/cortex-m-rt"

[dependencies.cortex-m-semihosting]
version = "0.2.0"

[dependencies.nb]
git = "https://github.com/japaric/nb"

[profile.release]
debug = false
lto = true
```

japaric, 2017-11-22:
OK. I ran into this today. The problem is subtle:

stm32f103xx, and other device crates, optionally depend on cortex-m-rt if the "rt' feature is enabled -- RTFM requires that "rt" feature to be enabled though.

Next, cortex-m-rt provides an implementation of the panic_fmt lang item *if* the "abort-on-panic" is enabled.

In your case, you have *two* cortex-m-rt crates in your dependency graph: one is pulled by stm32f103xx and it's a stable release (v0.3.6); the other one is the git version you have specified (git = "https://github.com/japaric/cortex-m-rt").

Your application is linking to stm32f103xx and thus is linking to cortex-m-rt *v0.3.6* because that's a dependency of stm32f103xx. However, that cortex-m-rt crate does *not* contain the panic_fmt lang item because its "rt" feature is not enabled. You have enabled the "rt" feature of the git version of cortex-m-rt but you are not linking to that crate.

Dependency management is fun, right?

TL;DR The fix is to replace `git = "https://github.com/japaric/cortex-m-rt"` with `version = "0.3.6"`. This is how the dependencies are written in the cortex-m-quickstart template.

If for some reason you do need to use the git version of cortex-m-rt and want to make this work then you'll have to add this to your Cargo.toml:

``` toml
[replace]
"cortex-m-rt:0.3.6" = { git = "https://github.com/japaric/cortex-m-rt" }
```

That ought to work.

MrBuddyCasino, 2017-11-22:
Thanks for the detailed reply, sorry for causing trouble by messing with the dependencies.
Btw., I recently tried the F3 Discovery board and that worked out of the box. Looking forward to exploring what you've created!

closed 2017-11-22 by MrBuddyCasino