japaric/stm32f103xx-hal #33: Build error

lnicola, 2017-12-19:
Am I doing something wrong?

```text
xargo build --release --example blinky         
   Compiling blue-pill v0.1.0 (file:///home/grayshade/blue-pill)
error: linking with `arm-none-eabi-ld` failed: exit code: 1
  |
  = note: "arm-none-eabi-ld" "-L" "/home/grayshade/.xargo/lib/rustlib/thumbv7m-none-eabi/lib" "/home/grayshade/blue-pill/target/thumbv7m-none-eabi/release/examples/blinky-9fb6fa70bf8cc687.aligned0-d85e69f32a001b7247982de9cdc16ee3.rs.rcgu.o" "/home/grayshade/blue-pill/target/thumbv7m-none-eabi/release/examples/blinky-9fb6fa70bf8cc687.bare_metal0-ceb19f83b9d1a417c53fc10e067f36c4.rs.rcgu.o" "/home/grayshade/blue-pill/target/thumbv7m-none-eabi/release/examples/blinky-9fb6fa70bf8cc687.blinky0-4a074f58195296a925d077a0b1de6b95.rs.rcgu.o" "/home/grayshade/blue-pill/target/thumbv7m-none-eabi/release/examples/blinky-9fb6fa70bf8cc687.blue_pill0-e0100c0169d57470faeb6a12630c98d9.rs.rcgu.o" "/home/grayshade/blue-pill/target/thumbv7m-none-eabi/release/examples/blinky-9fb6fa70bf8cc687.cast0-65f58988949b80eada6acccdc5021724.rs.rcgu.o" "/home/grayshade/blue-pill/target/thumbv7m-none-eabi/release/examples/blinky-9fb6fa70bf8cc687.core0-2ac42e5c3350e49ce90e504514ab12ee.rs.rcgu.o" "/home/grayshade/blue-pill/target/thumbv7m-none-eabi/release/examples/blinky-9fb6fa70bf8cc687.cortex_m0-84e15f74949370e151cbd47a201908a5.rs.rcgu.o" "/home/grayshade/blue-pill/target/thumbv7m-none-eabi/release/examples/blinky-9fb6fa70bf8cc687.cortex_m_rt0-1119d0553a8e541b9cc052e46f566d5d.rs.rcgu.o" "/home/grayshade/blue-pill/target/thumbv7m-none-eabi/release/examples/blinky-9fb6fa70bf8cc687.cortex_m_rtfm0-4de35aca160d7fbef9ab5e536e21a56e.rs.rcgu.o" "/home/grayshade/blue-pill/target/thumbv7m-none-eabi/release/examples/blinky-9fb6fa70bf8cc687.embedded_hal0-27a3320987144148e28022ccc4b488d9.rs.rcgu.o" "/home/grayshade/blue-pill/target/thumbv7m-none-eabi/release/examples/blinky-9fb6fa70bf8cc687.nb0-2b1664fbce6a938245fe80c1702d386d.rs.rcgu.o" "/home/grayshade/blue-pill/target/thumbv7m-none-eabi/release/examples/blinky-9fb6fa70bf8cc687.r00-3e174e1772c6e4b17978b00559c618ef.rs.rcgu.o" "/home/grayshade/blue-pill/target/thumbv7m-none-eabi/release/examples/blinky-9fb6fa70bf8cc687.rtfm_core0-8823343cdd451ea85048d97912dad459.rs.rcgu.o" "/home/grayshade/blue-pill/target/thumbv7m-none-eabi/release/examples/blinky-9fb6fa70bf8cc687.static_ref0-f771221c23600a6422236b89cd1aeedd.rs.rcgu.o" "/home/grayshade/blue-pill/target/thumbv7m-none-eabi/release/examples/blinky-9fb6fa70bf8cc687.stm32f103xx0-5b37a61009b9b23c28877807e94d7b44.rs.rcgu.o" "/home/grayshade/blue-pill/target/thumbv7m-none-eabi/release/examples/blinky-9fb6fa70bf8cc687.untagged_option0-b9ab34e49e380d14a8ef45d286f96c0c.rs.rcgu.o" "/home/grayshade/blue-pill/target/thumbv7m-none-eabi/release/examples/blinky-9fb6fa70bf8cc687.vcell0-529000b439629f6f3eb3daaf27051.rs.rcgu.o" "/home/grayshade/blue-pill/target/thumbv7m-none-eabi/release/examples/blinky-9fb6fa70bf8cc687.volatile_register0-6c42e7cb355346c5c447351b0be08d41.rs.rcgu.o" "-o" "/home/grayshade/blue-pill/target/thumbv7m-none-eabi/release/examples/blinky-9fb6fa70bf8cc687" "--gc-sections" "-L" "/home/grayshade/blue-pill/target/thumbv7m-none-eabi/release/deps" "-L" "/home/grayshade/blue-pill/target/release/deps" "-L" "/home/grayshade/blue-pill/target/thumbv7m-none-eabi/release/build/blue-pill-40cd7fce81af253c/out" "-L" "/home/grayshade/blue-pill/target/thumbv7m-none-eabi/release/build/cortex-m-rt-085da2a8930e8fcc/out" "-L" "/home/grayshade/.xargo/lib/rustlib/thumbv7m-none-eabi/lib" "-Bstatic" "/home/grayshade/.xargo/lib/rustlib/thumbv7m-none-eabi/lib/libcompiler_builtins-5e3db6aa2fee9b71.rlib" "-Tlink.x" "-Bdynamic"
  = note: /home/grayshade/blue-pill/target/thumbv7m-none-eabi/release/examples/blinky-9fb6fa70bf8cc687.stm32f103xx0-5b37a61009b9b23c28877807e94d7b44.rs.rcgu.o: In function `WWDG':
          stm32f103xx0-5b37a61009b9b23c28877807e94d7b44.rs:(.text+0x0): undefined reference to `DEFAULT_HANDLER'
```

This is on Arch Linux (`arm-none-eabi-ld 2.91`).

lnicola, 2017-12-19:
Works with `2017-12-02`.

japaric, 2017-12-20:
I thought this was fixed already. There was a small window (1 week?) of nightlies with this problem. cf. rust-lang/rust#46543. Can't check a recent nightly atm but I have nightly-2017-12-14 and this works fine. Does this happen with a recent nightly?

lnicola, 2017-12-20:
You're right, sorry. I tried searching https://github.com/rust-lang/rust/ for `DEFAULT_HANDLER`, but somehow missed that issue. It's working in the latest nightly.

closed 2017-12-20 by lnicola