japaric/stm32f103xx-hal #38: Clock source selection

nallar, 2018-01-25:
The CFGR builder API doesn't let you select a clock source.

Possible sources:

* HSI (high speed internal) oscillator
* HSI as source for PLL
* HSE (high speed external) oscillator
* HSE bypass
* HSE as source for PLL

Currently it only uses the first two options.

Should set cr.hseon and wait for cr.hserdy before setting up PLL. Also HSEBYP if using a clock source and not a crystal.
cfgr.pllsrc().external() swaps to HSE as pll source. Probably want pllxtpre().no_div().

Maybe like this?
```rust
    let clocks: Clocks = rcc.cfgr
        .sysclk(72.mhz())
        .hclk(72.mhz())
        .pclk2(72.mhz())
        .pclk1(36.mhz())
// This by default?
//       .source().internal()
        .source().external()
//      .source().external_bypass()
        .freeze(&mut flash.acr);
```

japaric, 2018-02-07:
> Maybe like this?

Looks good to me but I would make `external` unsafe since it requires the caller to pass the right value. The rationale of unsafe is that if the caller passes a wrong value that could result in a valid clock configuration (e.g. real system clock frequency of 64 MHz) but a wrong flash latency value (e.g. no latency because the value computed in software was 24 MHz) and that would result in undefined behavior (IDK the full consequences but I heard from someone that ran into this that the processor would jump to the hardfault handler at random locations of the program).

mvirkkunen, 2018-10-02:
@TeXitoi Yes, I did, but only after I sent that PR. There doesn't seem to be much activity on this repo at the moment so I'm leaving it be and focusing on other things until somebody wakes up.