japaric/stm32f103xx-hal #52: stm32f103xx-hal not available on crates.io

jamwaffles, 2018-02-19:
Unless I'm mistaken, this crate isn't published on crates.io. I'm currently including it in my project like this:

```rust
[dependencies.stm32f103xx-hal]
git = "https://github.com/japaric/stm32f103xx-hal.git"
features = ["rt"]
version = "*"
```

but of course it would be great both for ease of use and discoverability to make it available through crates.io.

Is it published and I'm missing something? If not, I think it's feature-complete enough to publish :)

therealprof, 2018-02-28:
Indeed it's not there which is also a reuse problem since you cannot publish a crate depending on a non-published crate.

@japaric Anything I can do to help get this published?

japaric, 2018-03-10:
> Indeed it's not there which is also a reuse problem since you cannot publish a crate depending on a non-published crate.

The same applies to this crate. As long as its dev dependencies are not on crates.io this crate can't be published either. We could comment out the dev dependencies / remove some examples just to get it published though.

I'd like to review the DMA API before publishing; I might put some of it behind an unstable Cargo feature gate if I'm still not too happy with it (e.g. the circ buffer stuff).

jamwaffles, 2018-03-11:
That's fair. I didn't know about the non-published dependencies limitation. It would be good for discoverability (particularly as doc/blog references to the Blue Pill crate are now dangling) and adoption. You could release early and keep updating things with breakages, but I understand wanting to hold back until the crate is more feature complete. Just as long as there is a plan to publish in the future! 😊

MajorBreakfast, 2018-03-29:
Will the `stm32f103xx-hal` crate eventually replace the `stm32f103xx` crate?

therealprof, 2018-03-29:
No, the `stm32f103xx` contains just the register descriptions for the MCU while the `stm32f103xx-hal` implements specific interfaces using the register description. I've actually started doing both in once but `svd2rust` really doesn't support that and I guess some people might enjoy banging away at the registers without having an HAL layer implemented.

MajorBreakfast, 2018-03-29:
@therealprof Thx for the clarification. I've now created an issue for this. I'd be good if there was a quick explanation about how the two crates relate in the `stm32f103xx` readme (https://github.com/japaric/stm32f103xx/issues/20)

rudihorn, 2018-05-09:
Could we push the extra examples into a further branch and remove them from the main branch? I could try to help out by submitting PR's for this?

Mendelt, 2018-08-25:
Any progress on this? It seems a bit weird to me to have so many dev dependencies just for the examples. Maybe some of the examples that depend on non-crates.io stuff should live in the repo's of the crate they depend on. For example the motor driver example could live in the motor driver repo?
Another solution could be to move some of the more extensive examples to one of the books.
This might also help in getting this crate to build with the stable toolchain.

mvirkkunen, 2018-12-02:
Are the example dev-dependencies literally the only reason this isn't on crates.io yet right now, or is there something else as well? I don't think it makes sense to make it cumbersome for everybody to use this crate just for the sake of a couple of examples.

therealprof, 2018-12-02:
@mvirkkunen That's it. You might want to check out #106 and register your voice there.