japaric/stm32f103xx-hal #76: MonoTimer does not work without debugger connected

TheZoq2, 2018-04-25:
My current project reads a `MonoTimer` in order to keep track of signals which have been received. These signals are then sent over serial to a pc. This works flawlessly when the stlink debugger is connected, however without it connected the value read is always 0.

I assume this is because monotimer uses the "debug watch trace" peripheral which judging by the name means that it doesn't work without a debugger. Perhaps this should be specified in the documentation, or perhaps mono timer could be rewritten to use something that works without the debugger.

One thing to note is that the frequency reading still works.

kellerkindt, 2018-09-01:
Having the same issue, but only works if it just has been flashed by the debugger. Unplugging, and plugging it back to the debugger causes the ```MonoTimer``` to always read 0 ticks elapsed, until reflashed.

kellerkindt, 2018-09-01:
@TheZoq2 Got it running with the following workaround, which I call before ```MonoTimer::new```:
```unsafe { *(0xE000EDFC as *mut u32) |= 0x01000000; }```
See example C Code [here](https://stackoverflow.com/questions/13379220/generating-nanosecond-delay-in-c-on-stm32) and [here](https://www.stm32duino.com/viewtopic.php?t=1057), especially this code section:

> #define DEMCR_TRCENA    0x01000000
> 
> /* Core Debug registers */
> #define DEMCR           (*((volatile uint32_t *)0xE000EDFC))
> ...
>  /* Enable DWT */
> DEMCR |= DEMCR_TRCENA;

TheZoq2, 2018-09-02:
Cool, I have since moved on to another solution without the `MonoTimer` but this might come in useful in the future.

A more rusty approach would be to use the `demcr` register here https://japaric.github.io/stm32f103xx-hal/cortex_m/peripheral/struct.DCB.html#structfield.demcr to avoid having a raw pointer. Might be worth making a pull request to fix it

kellerkindt, 2018-09-02:
> A more rusty approach

Haha, definitely. I just needed **some** kind of a solution yesterday evening :)
Interesting that you found the ``demcr`` register, although I searched for it with the github search. Might have missed cortex_m though.
I am still a bit lost in the rust abstractions...

tib888, 2018-09-08:
I just noticed that MonoTimer seems to be stopped, while I writing something to hstdout.