japaric/stm32f103xx-hal #88: definition of an unknown language item: panic_fmt

fmmrtns, 2018-06-18:
Hi! Thanks for this crate! 

I'm having a problem when building `cortex-m-quickstart` which adds this crate as a dependency:
```
> error[E0522]: definition of an unknown language item: `panic_fmt`
>   --> .../.cargo/registry/src/github.com-1ecc6299db9ec823/panic-semihosting-0.2.0/src/lib.rs:67:1
>    |
> 67 | #[lang = "panic_fmt"]
>    | ^^^^^^^^^^^^^^^^^^^^^ definition of unknown language item `panic_fmt`
```

Bumping `panic-semihosting` to version 0.3.0 should fix the issue, is there anything else I can do to fix this?

Thanks

fmmrtns, 2018-06-18:
Just saw this PR which fixes this issue: https://github.com/japaric/stm32f103xx-hal/pull/86 :+1:

japaric, 2018-06-28:
Fixed in #86

closed 2018-06-28 by japaric