japaric/stm32f103xx-hal #102: Entry point is not getting set for example programs

tsathishkumar, 2018-10-08:
I'm trying to build blinky example with default configuration. But it seems that the entry point is not getting set by the compiler.

Output of readobj:
```
➜  stm32f103xx-hal git:(master) ✗ cargo readobj --example blinky  -- -file-headers
    Finished dev [unoptimized + debuginfo] target(s) in 0.09s
ELF Header:
  Magic:   7f 45 4c 46 01 01 01 00 00 00 00 00 00 00 00 00
  Class:                             ELF32
  Data:                              2's complement, little endian
  Version:                           1 (current)
  OS/ABI:                            UNIX - System V
  ABI Version:                       0x0
  Type:                              EXEC (Executable file)
  Machine:                           ARM
  Version:                           0x1
  Entry point address:               0x0
  Start of program headers:          52 (bytes into file)
  Start of section headers:          2108984 (bytes into file)
  Flags:                             0x5000200
  Size of this header:               52 (bytes)
  Size of program headers:           32 (bytes)
  Number of program headers:         3
  Size of section headers:           40 (bytes)
  Number of section headers:         16
  Section header string table index: 14
```

Output of size:
```
➜  stm32f103xx-hal git:(master) ✗ cargo size --example blinky --release -- -A
    Finished release [optimized + debuginfo] target(s) in 0.14s
blinky  :
section              size   addr
.debug_str          39258    0x0
.debug_loc           4323    0x0
.debug_abbrev        1993    0x0
.debug_info         40641    0x0
.debug_ranges        6200    0x0
.debug_macinfo          2    0x0
.debug_pubnames     11026    0x0
.debug_pubtypes     13001    0x0
.ARM.attributes        50    0x0
.debug_frame         1508    0x0
.debug_line         17848    0x0
.comment               18    0x0
Total              135868
```

TeXitoi, 2018-10-08:
We need a repo to reproduce your problem.

Try this: https://github.com/TeXitoi/blue-pill-quickstart

closed 2018-10-08 by tsathishkumar

TeXitoi, 2018-10-08:
No, it depends on the version of cortex-m-rt that you are using in your Cargo.toml. The examples on this repo use 0.5, and blue-pill-quickstart use the last version 0.6.

tsathishkumar, 2018-10-08:
Ok, so 0.5 is not working I guess. I'm going to stick with 0.6 for now. Thanks for the quick replies.

tsathishkumar, 2018-10-08:
I didn't change the version or `cortex-m-rt`. It is 0.5.2, and the entry macro is used in the example. But it doesn't work (at least in macos).

TeXitoi, 2018-10-08:
I can analyze this if you give a full project, but as you have something working, that may not be important.