japaric/stm32f103xx-hal #108: Constrain AFIO leads to unsafe code when setting up EXTI

freiguy1, 2018-10-19:
I'm working on a little hobby project on a blue pill with rtfm and this lib. I just added some PWM to the project which required calling `constrain` on AFIO.  A few lines after constraining, I want to do something like this:

```rust
p.device.AFIO.exticr3.write(|w| unsafe { w.exti11().bits(0x01) });
```

To enable EXTI11 on port B. Constrained AFIO does not have EXTICR3 available, so I tried it this way. However, AFIO has been _moved_ so this line doesn't work and instead I need to do the `unsafe { &*AFIO.ptr() }` bit to get access to it. 

I'm striving to minimize unsafe code. Is this a thing that will eventually be implemented on afio::Parts? Or is there a better way I can be doing this?

therealprof, 2018-10-20:
@freiguy1 This can be easily solved by adding `exticr3` (and similar registers as well) to `src/afio.rs` like it is done with `mapr` in there. Would you like to take a stab at it and submit a PR?

freiguy1, 2018-10-20:
Yes, that could be a great way to get my fifth [Hacktoberfest](https://hacktoberfest.digitalocean.com/) pr as well :) I'll work on it

freiguy1, 2018-10-20:
@therealprof Question:
MAPR struct looks like this currently:

```rust
pub struct MAPR {
    _0: (),
}
```
I'm confused why it needs the _0 field. It seems to build like this: `pub struct MAPR;` Maybe there's an inconspicuous reason why it's required.

TeXitoi, 2018-10-21:
That's to have a private field to forbid construction by the user.

therealprof, 2018-10-21:
@freiguy1 What @TeXitoi said.

freiguy1, 2018-10-22:
PR is available: #113

closed 2018-10-22 by freiguy1