japaric/stm32f103xx-hal #120: Adding support for abitrary dma transfer

aurabindo, 2018-12-07:
At the moment DMA can be done with peripherals. One cannot use the hal to configure the dma subsystem to have arbitrary dma transfers, or transfers triggered by Timer peripherals. How about adding some configuration api to Channels in addition to `listen` and `unlisten` ?

therealprof, 2018-12-07:
@aurabindo You mean as an extension of the `embedded-hal` traits? Sounds like a good idea, feel free to propose it over at: https://github.com/rust-embedded/embedded-hal

aurabindo, 2018-12-07:
@therealprof to add it to embedded-hal, we need to analyze how other microcontrollers do it as well for arriving at a common ground right?

therealprof, 2018-12-07:
@aurabindo I didn't mean to suggest this should be closed. I think it's a reasonable request and will certainly require some implementation in this crate.

aurabindo, 2018-12-08:
I just looked at the embedded hal crate, and it doesnt have anything related to DMA. However, this crate and many other do have an implementation. Does it mean it is beyond embedded-hal's scope?

therealprof, 2018-12-08:
@aurabindo It just means no one has come up with a clever idea to design those traits yet.